Builty/TonexOneController #188: Minimidi binary

Slowspark, 2025-05-17:
Can I find the bin files for minimidi somewhere?

Builty, 2025-05-17:
Hi. The minimidi was a quick project that I did, but then developed the WiFi control method and didnt use it any more.
I can build the current version for you if you like, but I don't have releases like the main project, and don't plan to develop it any further.

Slowspark, 2025-05-17:
I would be grateful for build because I have a problem with installing esp idf, it shows some errors.

Builty, 2025-05-19:
I haven't tested this one, but the code is unchanged from last time I did.
I don't think I have the hardware still assembled to run it. If you run into issues, I can check more carefully.

[minimidi_V1.0.0.2.zip](https://github.com/user-attachments/files/20280726/minimidi_V1.0.0.2.zip)

Slowspark, 2025-05-19:
Thank you very much, I uploaded it and it works with 2 up and down buttons, I have a question, how to configure it to work with 4 buttons?

Builty, 2025-05-19:
[minimidi_V1.0.0.2_4button.zip](https://github.com/user-attachments/files/20306850/minimidi_V1.0.0.2_4button.zip)

For this one, it needs the code change (see below). I never got to adding WiFi or anything like that. Here is a build.

[image]

Slowspark, 2025-05-27:
Thank you very much, it works, it would be useful to add 4 LEDs and signal the last pressed button, but since you are not developing it anymore, maybe I will add it myself someday