Builty/TonexOneController #193: Unable to connect to tonexconfig / tonex.local

josemejiazzz, 2025-05-19:
Greg,

I was able to set up the controller but I don't have access to the web control. I tried to look for the tonexconfig network and also tried to go into tonex.local. I was able to do it last night once but I can't do it anymore. Any thoughts? I was also able to install the 1.0.10.2 version.

Great job, Greg!

Jose.

Builty, 2025-05-19:
Hi Jose.
The access point that is used for configuration is enabled for 1 minute after boot. This can be changed in the web config to be on all the time if you need.
Try rebooting the controller via removing its power, and then wait a few seconds for it to start up, then use a phone/tablet/PC with WiFi to scan for the access point.

josemejiazzz, 2025-05-20:
Greg,

I have done that several times with different devices: computer, two different phones and it doesn't work. I've tried to reset the controller, re-installed the previous version to see if it helped... nothing.

Thank you,

Jose.

Builty, 2025-05-20:
Hi Jose. Did you adjust any of the settings in the wifi area?
I would suggest using the programming application again, but this time press the Erase button. Wait for it to finish, and then program it again normally like you did before.
This will clear out the configuration.

Then try the access point again. What I do is switch on power, wait a few seconds, and then use my phone to scan. It takes a little time for the phone to see it. Some phones allow you to drag down on the screen to refresh the scanning.

josemejiazzz, 2025-05-21:
Greg,

I can't access the app nor the access point. It doesn't show on the available networks. I have re-installed the flash and tried every combination of protocols: powering up the ESP32 and try to find the network, plug, unplug, boot, etc. I can't find the network. Without the network, I have no access to tonex.local

Any thoughts?

Jose.

josemejiazzz, 2025-05-21:
Greg, 

I think I understand what you are saying. Please disregard the last message. I did what you told me to do and it worked. I’m going to try to set it up now. 

Thank you!

José

josemejiazzz, 2025-05-21:
Greg,

I'm able to connect now but, if I change any parameters on the webcontrol, it doesn't change anything on the controller. Also, I can't change the presets on the controller by pressing the arrows next to the effects buttons. I was able to use the BT chocolate device and it doesn't work anymore. I can definitely connect to the controller but it doesn't really sync with it.

Any thoughts?

Thank you,

Jose