Builty/TonexOneController #194: Power up issues on ESP 32 zero

Gilbertnghk, 2025-05-20:
I am using V1.0.10.2 Beta 2 with ESP 32 zero 
I tried  using the adaptor 

USB C OTG Adapter USB C to USB 3.2 Converter Adapter with 100W

Or connecting the chip as your instruction mentioned. But when it is connected to power bank with PD feature,  the chip seems couldn't get enough power,,it suddenly turned off or couldn't power up my tonex.  

is there any solution? Please help. Thank you.

Builty, 2025-05-21:
Hi. A few people have had trouble using USB power supplies. Even though the battery supports PD, the controller does not, so it can't take advantage of the higher current.

I think its best to start using a normal 5v DC "plugpack" type of power supply, connected to the 5v input on the Zero. Just to make sure that the rest of the system is OK and no other issues exist.

I don't use this kind of adaptor or power bank myself, so its not easy for me to provide a lot of advice.

dietzo, 2025-05-24:
Hi. To solve this issue (PD power bank not powering my ESP32 zero and Tonex One), I added a USB-C PD trigger/voltage converter between my USB-C PD power bank and the ESP32 zero. The voltage converter is a small PCB with USB-C port that can be set to trigger 5V from a PD power bank. I wired/soldered the voltage converter's + and -  to the ESP32 zero's + and -, respectively. Then I connected the power bank via USB-C cable to the voltage converter's USB-C port which will then power the ESP32 zero with 5V. I connected the ESP32 zero via USB-C cable to the Tonex One, allowing to power the Tonex One and use the ESP32 zero for MIDI functionality. Hope this helps and works for you/others as well!