Builty/TonexOneController #197: Unable to use M-Vave Chocolate Plus and unable to "Sync" web control with controller

josemejiazzz, 2025-05-21:
Hello Greg,

I was able to connect the controller and my computer/phone/tablet. I also changed the parameter that allows the controller to be visible all the time instead of just one minute. I can make changes to some things but there's a lot of other things I can't change/sync. The issue I'm facing now is that I can't change presets by pressing the arrows next to the effects buttons on the controller. When I make changes on the webcontrol, they don't reflect on the controller, as you showed on your video. I can't use the BT chocolate plus pedal. I was able to use it two days ago but now I can't. The version that I installed is the 1.0.10.2 (beta).

Any thoughts?

Thank you,

Jose.

Builty, 2025-05-22:
Hi Jose.
Are you running the latest firmware on your Tonex pedal? About 1 month ago IK changed some that caused issues like you descibe.
Please open the Tonex app, and update it, then update your Tonex pedal to the latest.

josemejiazzz, 2025-05-22:
Greg,

You are legend!!!

I updated the ToneX One pedal and it worked!!! I was getting the ToneX PC version with the Editor. I finally know the difference!

Great work!!!

Jose.

Builty, 2025-05-22:
Good to hear. This issue has affected a few people. I should note with Tonex firmware is compatible.