Builty/TonexOneController #211: M5Stack Atom S3R not working

greho, 2025-05-29:
First: thanks for this awesome project!
I got my M5Stack Atom S3R device yesterday. I did all the steps as explained (several times)and flashed the firmware to my Atom.
Now the screen is dark, nothing happens. I can power my Tonex One (Atom into my power supply 5V, USB C to Tonex), but I can do nothing else. No Wifi or Bluetooth will be detected in my Phone/i Pad/ Macbook devices... Has anyone any idea what went wrong?
By the way: it's a new M5Stack Atom S3R. My Tonex Firmware is the latest (1.3.14). I tried several releases (1.08.0.2 to 1.10.0.2) on Windows and on Mac (my usual platform). Transferring always showed up as finished. But my M5Stack Atom S3R seems "dead".

andy-k-dot, 2025-05-29:
I have one working fine on 1.09, and latest Tonex firmware.
Only things I can suggest, is, try re flashing but use the erase function first, and make sure it finishes.
If it is detected by your pc, and you get that far, it should be working.
Secondly, power can be tricky, I am powering mine from a USB powerbank, which is 5v, 3 amp, and I can have issues when the powerbank drops below a certain level.
I was having problems with a powerbank that was putting out 5v, 2.4 amp, but somehow that seems to work fine now?
Does the Atom work on its own? - before you connect Tonex?
Does Tonex seem to power up?, if it is flashing there is not enough power.
There is not really much more I can suggest, but first step is to get it flashed correctly, and check power supply is correct.
good luck

Builty, 2025-05-30:
Are you 100% sure its an S3R? 
M5Stack make S3 and S3R. 
They look the same but only the R will work.

greho, 2025-05-31:
Dear andy-k-dot and Builty,
thanks for your remarks and your time.
First my reply to Builty:
it‘s an S3 - no „R“… Maybe that‘s the solution. I ordered another one with S3R.
I‘ll give an update in a few days.

And to andy-k-dot:
the atom worked by the first plug in. After I gave it the tonex midi firmware it‘s „dead“.
It powers up my Tonex with no issues.
I don‘t have the ability to measure the exact Voltage, but I tried several powering methods including 2 different powerbanks. Nothing changed.
But I give your suggestions another try and update you.
But maybe the problem is the wrong device (as mentioned above)?

Greetings
Gregor

Builty, 2025-05-31:
Hi Gregor. The S3 (no R) is not the same hardware as the R. It won't work, as you can see. Only the S3R is supported.