Builty/TonexOneController #235: ESP32-S3 Zero bluetooth peripheral mode does not work (with iphone/ipad)

Fozziemusic, 2025-07-10:
In an otherwise well-functioning Zero setup, I have tried to connect my iphone and ipad to connect to the Zero using Bluetooth in 'peripheral' mode. First I was on firmware 1.0.9.2, just updated to 1.0.10.2, but the result is the same:
- 
- in 'central' mode my Zero connects with a Chocolate perfectly
- in 'peripheral' mode I cannot get Iphone or Ipad to see the Zero via bluetooth. It may not broadcast or in an unsuitable way for IOS devices to 'see' (not sure if that can be the case).

This issue was the same in both firmware versions. And I have tried all options - using custom BT device with my Iphone name, with midi cc enabled/disabled, same for Chocolate and MD1 options. Am I missing something?

I am looking for ways to send midi from an iphone or ipad to the Zero/tonex one. Via the chocolate it works, via web config changing settings works, but no midi from iphone/ipad so far.

PS: I hope I'm not posting too much - please let me know if so.

Fozziemusic, 2025-07-10:
Ok, this one I was able to solve myself after having searched the net already for a long time. 

In IOS bluetooth settings/connections the bluetooth connection from Zero does not show up. HOWEVER, with the **app 'Bluetooth inspector'** more bluetooth connections can be seen and connected with, including TnxBT!! After connecting to it in the app, the BT connection now shows up in my regular BT IOS setting.

closed 2025-07-10 by Fozziemusic

Builty, 2025-07-11:
Hi. Its fine to post a lot of details, actually I much prefer that.

I'm not an iOS user so I have no idea about its bluetooth handling, but I'm glad you solved it and hopefully the info can help other people.