Builty/TonexOneController #237: How to power Waveshare 1.69 external footswitches?

sergesafo, 2025-07-20:
Hi! I've got all components for Waveshare 1.69 external footswitches incl. SX1509 and my question is: how to power this circuit? Does it require power supply? I can't see it on this schematic

[image]

sergesafo, 2025-07-20:
> This version doesn't need the extra board for 4 switches, and needs no additional power. There are already enough connections for 4 switches, but if you need up to 16, I guess you would use the SX1509. I had to use the extra board for my 4.3b build, as that only has connections for 2 onboard switches, 1.69 works fine with 4. Power supply is the red and black connections to the SX1509.

andy-k-dot thank you for answering. You know, I still don't get it. Let me explain. I'm going to use Waveshare 1.69 alongside with SX1509 and 10 footswitches. If I got it right SX1509 will supply power to Waveshare 1.69, correct? And Waveshare will supply power to Tonex One via USB?

Builty, 2025-07-20:
Hi. The SX1509 is powered off 3.3v. The 3.3v comes from the Waveshare 1.69 board, the red wire.
Usually you would connect 5 volts to the Waveshare board. The Waveshare board will provide 5v power on its USB port for the Tonex. And the Waveshare board converts 5v to 3.3v, and that 3.3v then powers the SX1509.

Andy K is correct in that if you wanted 4 switches, you wouldn't need the SX1509. But to have 10 switches then yes you do need it.

sergesafo, 2025-07-20:
> Hi. The SX1509 is powered off 3.3v. The 3.3v comes from the Waveshare 1.69 board, the red wire. Usually you would connect 5 volts to the Waveshare board. The Waveshare board will provide 5v power on [...]

Builty, thank you for answering. So I still need to connect 5 volts to the Waveshare board? As it's shown on this picture 

[image]

Builty, 2025-07-21:
yes thats correct. This is the only power supply needed.

sergesafo, 2025-07-21:
Builty, thanks a lot!

sergesafo, 2025-08-10:
Hi! So I made all connections (attaching the picture), exactly like on the schematics, uploaded latest firmware to WS 1.69 (v2.0.0.2), made  2X5B Preset Switch Layout  via Web Control, but, unfortunately, no I'm not able to switch presets. What I've done wrong?   

[image]

sergesafo, 2025-08-11:
Yes, I do

Builty, 2025-08-12:
I ran tests on V2.0.0.2 on my 1.6". Its working OK for me.

Looking at your photo, are the SCL/SDA/3v3 pins connected? The diagram shows the top side of the SX1509 board. You have some solder on RST and INT, which aren't needed. Please check the connections here.

[image]

sergesafo, 2025-08-12:
Hi Builty, so I decided to make everything from scratch and, finally, it worked out for me. And I want to thank you for this wonderful project. It will make my gig life so much easier. I want to express my deep appreciation for your tireless work on this project and for your help and support!

Builty, 2025-08-16:
You can see the wiring for 4 buttons here:
https://github.com/Builty/TonexOneController/blob/main/HardwarePlatforms.md#footswitches

That SX1509 PCB looks OK.
When you had 10 switches connected, did you change the External switch mode in the config to a layout that uses them all?
I have 10 footswitches on my pedalboard, and 16 on a test system, and they all work OK for me.

Builty, 2025-08-16:
I searched for this board and found it on Aliexpress.
In the top view image, the numbers don't seem to match the bottom view.
I think the issue is something to do with this PCB. Can you check both top and bottom and make sure the right numbers are used?

[image]

Builty, 2025-08-16:
Actually I can see the PCB traces to the chip pins. It does look to be OK and consistent from front and back.
I'm not sure what the issue would be.
I'm going to order one myself.

sergesafo, 2025-08-17:
Actually, for the recent project I use the red SX1509 PCB.

Builty, 2025-08-17:
Can you please double check the wiring, to make sure no shorts between pads, and that all wires go the same spot.
Maybe post a photo of the complete wired PCB?

Builty, 2026-01-06:
Hi, that's great.
I've had a few requests for expression pedal support. I do plan to do it, but I don't have a time line at the moment. I'm trying to crush the last few pesky bugs in V2.0.2.2 before I start up on anything new.