Builty/TonexOneController #238: Wired Midi questions (Atom S3R)

hardys, 2025-07-22:
Firstly thanks for the great project!

I have it working well on an Atom S3R running release version 1.0.9.2, however I'm unable to make the wired MIDI work (enabled via web UI).

I have a Midi FeatherWing attached as described in [the docs](https://github.com/Builty/TonexOneController/blob/main/HardwarePlatforms.md#midi), and verified the wiring etc is OK using a modified version of this [MIDI monitor](https://github.com/diyelectromusic/sdemp/blob/main/src/SDEMP/ESP32SimpleMIDIMonitor/ESP32SimpleMIDIMonitor.ino) - adjusted to use GPIO pins 38/39 as required for the Atom S3R.

I can see the following serial output which I think proves the wired MIDI connection to the S3R is working - we can see the CC messages on channel 1, 127 (Select Preset) - I'm trying to select presets 7/8 in this example.
 
```
1	0xB0 CtrlCh	0x7F	0x08
1	0xB0 CtrlCh	0x7F	0x07
``` 

Uploading the 1.0.9.2 firmware, and continuing to monitor the serial output we see
...
...
(1761) app_main: Init MIDI Serial
(1770) app_main: Init USB
(1770) app_midi_serial: Midi Serial task start
(1774) app_usb: Installing USB Host Library
```

Sending the same MIDI messages however do not result in changing preset on the ToneX one, any hints on how best to debug?

Understandably when the USB port is used for host/OTG it can't also be used to monitor log messages, and I've not yet got ESP-IDF configured to prepare a custom build - are there any other options to debug this?

Many thanks for any pointers you can provide!

Builty, 2025-07-23:
Hi,
Have you enabled wired midi in the web config? Also make sure the Midi channel is set correctly to match your config (default is channel 1, but can be changed in the web config.)

For this kind of thing, I suggest we try to identify if the issue is Midi, or if its general communications to the Tonex pedal.
If you change presets using the controls on the pedal itself, does the preset name on the screen change? 

If yes, USB comms is probably OK and it would look to be a Midi issue.
if no, the its probably a USB comms issue. In this case, make sure the Tonex is running the latest firmware, and make sure the USB-C cable you are using supports communications (sometimes called "charge and sync") and not just charge only.
They are the 2 most common causes of a lack of USB comms.

In terms of debugging in general, here are some tips
- if you can build the firmware, temporarily comment out the usb_comms_int() in main. This will kill Tonex comms, but will then allow viewing of the complete log, and when a Midi command is received the log will show what happens
- ESP32-S3 chips also send the logs on the "UART" pins (usually marked as TXD) which can be used to view logs even when the USB port is busy. Some platforms expose the externally, but it looks like the Atom doesn't. In the schematic R13 connects to GPIO43, but not sure where it goes from there. You could possibly attach a wire to that resistor, and then connect the other end to a USB-TTL UART adaptor cable, on its RX pin. That would allow monitoring of the logs via a terminal app.
I do this on some platforms myself.

hardys, 2025-07-25:
Thanks @Builty for the information - the USB comms seem OK, if I put the ToneX one in preset browsing mode I see the corresponding preset change on the Atom display.

I currently can't see how to connect to GPIO43, so I'll get the build environment working and try to debug with the `usb_comms_int` method mentioned.

I'll close this but many thanks for the help, and I'll report back when I figure out why this isn't working :)

closed 2025-07-25 by hardys