Builty/TonexOneController #241: Clarification about expression pedal not supported

erich74, 2025-07-28:
Hi, first off I would like to thank you for your work, what you have developped and shared is just awesome!
Just got my Tonex One and came across your project, I had a Zero board in stock so I quickly flased and paired it with my new Tonex One and Chocolate board.
So far so good. I would just like to clarify the point about expression pedal not (yet?) supported.
I have enabled midi cc in the web config tool and assigned an expression pedal connected to the Chocolate on CC103 in order to control the amp volume and it is working fine. I am then wondering if you mention an expression pedal directly soldered on the ESP which is not yet supported?

BTW I have read on your github that midi CC should not be enabled with the Chocolate but I read it after testing it and I have not noticed any problem with my patches. Do you have more info about what is altered when you enable midi CC with the chocolate? Do you plan to fix it? 

Thanks a lot!
Eric

Builty, 2025-07-29:
Hi Eric.
Yeah the expression pedal comment was in relation to directly connecting the analog part to an ADC on the ESP32. Some users have requested that, but not done so far.
As you can see however, its fine if the analog is converted to Midi by some other device.

With the Chocolate, when used in a standard/default mode for switching presets, when you change banks on it there is a CC message sent that triggers unexpected behaviour, hence Midi CC BT is disabled by default (a lot of users would be affected by this.)

I can't recall the CC index but I think it triggered a parameter adjustment.

If you use the Chocolate in some other mode, like just sending CC directly, then its all fine.
I couldn't really see any way of fixing it without breaking Midi compatibility with the big Tonex, which I wanted to avoid doing.

erich74, 2025-07-29:
Thanks a lot for your answer.
I am happy with the Chocolate default mode as I just want to send PCs for changing presets. 
I have not changed banks in my testing reason why I have not noticed any unexpected CC modification.
I will check if using any advanced mode of the Chocolate I can still use PCs (If I can get at least 2 banks of 4 presets it would be perfect)

erich74, 2025-08-04:
Hi again, just to be sure before I close the ticket: There are 2 default PC modes on the Chocolate: Program Change A and Program Change B. The first one has 8 banks of 4 PCs and the second mode has 32 banks of 4 PCs.

Can you please confirm whether the CC bug occurs on both factory banks or only the first one ?

The first mode is more convenient to accomodate the 20 presets of the Tonex One but if the second bank is working fine it is better than nothing

Thanks!

Builty, 2025-08-09:
Hi. Sorry for the late reply.
I'm not sure about the CC bug on different banks. The Chocolate pedals have lots of options and I don't use mine much (I only got it to do testing of the code.)

If you could show me some screen shots of what to change on the Choc, I can run some tests to see if the CC bank select applies to both.

alvikagal, 2026-01-21:
Is it possible to activate CC commands for the Waveshare Zero? I would like to try several functions: turning effects on and off, selecting effects, and tapping the TIME parameter for Delay.
If you can tell me where to activate it, I’m ready to test it on the M‑VAVE Chocolate Plus
Thank you.