Builty/TonexOneController #242: "Cheap Yellow Display" ESP32-S3 8048S043

erich74, 2025-07-28:
Hi all,

Just wondering if someone has already (or is currently) made this project work on the  so called Cheap Yellow Display ESP32-S3 8048S043 device that is widely available on Aliexpres at a reasonable price ?

Thanks!
Eric

Builty, 2025-07-29:
Hi Eric. Would you have a direct link to the product? I searched on Aliexpress but only got one hit and it was more expensive than a Waveshare.

For info, the next version has support for the JC3248W535C which is a similar board to the CYDs.

erich74, 2025-07-29:
Yes sure here is a link (this is the 4.3 with capacitive touch). It is often sold under Sunton brand and they have a wide range of touch/non touch resistive/capacitive models of various sizes. I also got the 7' model.

https://www.aliexpress.com/i/1005006112914604.html?gatewayAdapt=glo2fra

I agree prices have increased lately. I paid 23€ a couple of years ago.

Do you think this model would be cross compatible with the JC3248W535C  ? 

Thanks!

Builty, 2025-07-29:
Hi Eric,
That one is quite different to the JC3248W, but it is fairly close to the Waveshare 4.3B. It would still need to be a new platform as it uses different pins, and doesn't appear to have the same IO expander as the Waveshare.

In the last month I've added 4 new platforms, and its ended up costing a lot of time. If the CYD was much cheaper I'd be more keen to support it, but being so close in price to the Waveshare 4.3B, which can also come with a really nice plastic case, I think I'd recommend to people just to get the Waveshare.