Builty/TonexOneController #243: ESP32 LCD 4.3B Board Version.

paul0-andre, 2025-07-29:
Hi, could you tell me which of these versions I should install on my ESP32 LCD 4.3B Board?

1. Releases/V1.0.10.2/TonexOneController_V1.0.10.2_Waveshare_4_3B_Display_Pedal_Skins.zip

2. Releases/V1.0.10.2/TonexOneController_V1.0.10.2_Waveshare_4_3B_Display_Amp_Skins.zip

If I have to install both versions, which one do I install first?

thanks!

erich74, 2025-07-29:
Hi, I guess you have to choose one of them (you can't upload 2 different firmware) depending on if you want to display pedal or amp skins

Builty, 2025-07-29:
Hi. You caan choose one or the other. The "pedal skins" version has pictures of pedals in it, and the "amp skins" one has pictures of amps in it. The pictures were too large to include both options in the same firmware build.

paul0-andre, 2025-08-01:
I set up the Wi-Fi in "Station Mode" and now I can't connect to the card's Wi-Fi. The "tonex.local" network doesn't appear.

paul0-andre, 2025-08-01:
I used the programming tool to erase and reprogram the app, and it worked again. Thank you so much again!

paul0-andre, 2025-08-05:
Hi Greg! I'd like to ask you a question. I bought the Sparkfun SX1509 board and three footswitches to make a controller similar to the Pirate Midi - the "Max Controller." They recommend 600mA for the "Max Controller" with two footswitches. In my case, where I plan to use three footswitches, how much mA would I need?

Builty, 2025-08-05:
Hi Paul. 
The Polar Max uses the onboard inputs for the footswitches.
Adding the SX1509 will a small amount to the current draw, but its only about  1 mA. So small you can ignore it.
The same recommendations as the Max should be fine.

Builty, 2025-08-05:
I think you could try a single 500 mA and see how it goes. It should be close. Keep an eye out for any signs of power issues, such as leds getting dim or the controller rebooting.

Builty, 2025-08-09:
Hi. You can move banks up and down. Press button 1 and 2 at the same time, or 2 and 3 at the same time. This will change up or down banks.
If you prefer, you could add 2 more switches for dedicated up/down without needing to press two at once.

Make sure you have the right setting in the web config also. The "ext" tab has options for the footswitch layout.

paul0-andre, 2025-08-09:
Greg, would this type of foot switch work perfectly? Or can you recommend one with an LED so that the moment I step on it and press the foot switch, it turns on the light.

[image]

Builty, 2025-08-09:
The switch part itself would be fine. The led part could work, but it would need some wiring to make it switch on like that.
It looks like the switch has multiple "poles" (switch contacts.)
Use on pole for the input to the controller.
Use the other pole for the led. Connect power to one side and the led to the other. It might need a current limiting resistor also.
That should do the trick.

paul0-andre, 2025-08-30:
Hi Greg, my card isn't syncing with Tonex One. I reset the board and reinstalled the latest version, but it's still not working. Is there anything wrong with updating Tonex One?

[image]