Builty/TonexOneController #264: Add option to avoid forcing Stomp/C on preset load (enable A/B Dual mode with MIDI PC)

lucastephann, 2025-09-01:
Hi! First, thanks for the amazing project 🙌

**Use case**
I’m using TONEX ONE live with two different captures of the same amp per HX Stomp preset: one clean and one high-gain. I want MIDI Program Change from my HX Stomp to select the preset but keep the TONEX ONE in Dual (A/B) so I can use the pedal’s footswitch to toggle between those two captures on the unit itself.

**Current behavior**
When I send plain MIDI Program Change via the controller, the pedal is forced into Stomp and the preset is loaded into Slot C. That prevents the unit’s FS from cycling A/B between the two captures.
* Maintainer notes: wired MIDI uses Bank/Slot C always to guarantee activation.
* Hardware docs also mention the controller relies on Stomp mode and auto-sets it.
* There’s an older issue about forcing Stomp mode.

**Why this is an issue**
* In Stomp mode, I’d always have to “burden” one or two precious footswitches on my HX Stomp just to switch between the two captures. Meanwhile, the TONEX ONE footswitch is left aside, useless (no point in bypassing when it’s in my FX loop).
* I also lose access to the TONEX ONE tuner in Stomp mode, which forces me to dedicate another HX Stomp footswitch just for tuner access.
* This makes live operation less efficient and removes two core benefits of the pedal: A/B toggling and tuner.

**Expected behavior**
A configuration option to respect the pedal’s current play mode (Dual) and/or a way to explicitly target A or B when loading a preset via MIDI, so the unit’s FS continues to toggle A↔B and the tuner remains available.

**Why this matters**
For many players, A/B is the core live flow: two captures (clean and high-gain), stomp to switch, and tuner access when needed. Forcing Stomp/C removes both the A/B toggle and the tuner, which makes live use much less practical.

**Environment**
* TONEX ONE firmware: latest available
* HX Stomp firmware: latest available
* TonexOneController: latest available release
* Connection: HX Stomp sending MIDI PC → TonexOneController → TONEX ONE
* Repro: Send PC for preset N → pedal switches to Stomp/C → FS no longer toggles A/B, tuner unavailable.

Thanks for considering! Happy to test any beta or supply logs/videos.

madejm, 2025-09-02:
In Wifi Config under "Miscellaneous" tab, there is an option "Changing preset saves to slot:". Would this work for your case?

Builty, 2025-09-02:
I understand the issue. 
Tuner is actually still available in Stomp mode, via long press.

Madejm's suggestion might help. Its not easy via Midi, as there isn't a way I can think of to target A/B via Midi when selecting a preset. Need to think some more about this one.

madejm, 2025-09-03:
Could this be solved with different midi channels? Using channel 0 could change preset in active slot (as it does now), and channels 1-3 would set preset in slots A/B/C?

Builty, 2025-09-05:
I see your pull request. I'll check it in more detail soon and get it merged. Thanks for this.

Also that's interesting about the CI compile. I hadn't really thought about that, but yes it would be an easy way of compiling without having to setup a local install. I'm hoping to expand the CI to get it to create the zip releases.

lucastephann, 2025-09-06:
Yeah! It helped a lot, didnt had to set up anything in my environment, flashed straight from the browser, made everything easier!

lucastephann, 2025-09-09:
Solved in #266