Builty/TonexOneController #271: JC3248W535 BOARD HELP

deguitarman, 2025-09-14:
Hello, I'm looking for help regarding the wiring diagram of the JC3248W535 unit. I want to create a midi controllable unit with Feather wing.

1. What is the use of the TTL? It is not even connected? 

Should I just follow the wiring diagram without the TTL?

[image]

2. When flashing the firmware what is the one I will get? Esp32-s3 / Factory / Usb ?  I also clicked the left small button on the JC3248W535 to enter debug to flash firmware is this correct?

Builty, 2025-09-14:
Hi.
1. Sorry this was my mistake. That blue PCB was not supposed to be there. I have removed it now. Follow the diagram as shown with the Featherwing to the JC board.
2. This board was added in V2.0.0.2 so you need to use the beta here: https://github.com/Builty/TonexOneController/blob/main/Releases/beta/TonexOneController_V2.0.0.2_beta_2_JC3248W_Display_Amp_Skins.zip
3. The firmware upload procedure is covered here: https://github.com/Builty/TonexOneController/blob/main/FirmwareUploading.md

For the boot button, see here: https://github.com/Builty/TonexOneController/blob/main/datasheets/JC3248W535%20Specifications-EN.pdf

[image]

Builty, 2025-09-15:
Yep 5th pin is correct. Refer here:
https://github.com/Builty/TonexOneController/blob/main/datasheets/JC3248W535_GPIO_pinout.jpg

Pin 5 is GPIO16.

And in code:

[image]

deguitarman, 2025-09-15:
it's not work huhu Even tried switching Midi Type a or b cables. 

https://github.com/user-attachments/assets/3d7a921b-7101-4590-913b-8c0fa5b11549

Builty, 2025-09-15:
Oh, the the issue may not be with the config change.

Below is my horribly messy wiring (I have about 10 platforms to support and too lazy to make them all neat.)
Maybe something stands out as different?

[image]

[image]

[image]

deguitarman, 2025-09-16:
do you see anything wrong here? 

[image]

deguitarman, 2025-09-16:
[image]

looks okay to me.. ?

deguitarman, 2025-09-18:
i just made this midi din to type A
multi meter confirms connection
no avail

[image]

😭

Builty, 2025-09-20:
Hi. I'm sorry, its really hard to know what the issue is here. I think my code is correct.

I don't suppose you have an oscilloscope or logic analyser? If I had this issue I would be monitoring the output of the featherwing board to see whether it has the asyncronous serial midi data coming out from it. That would then verify the hardware.

deguitarman, 2025-09-22:
thanks for getting back. Ill confirm this week, got a midi to midi wires and will try it on an hx effects. ill try.let you know

deguitarman, 2025-09-27:
also @Builty May I ask what is expected Midi values the controller ? So its Midi Channel, Midi PC that's it? No CC value correct?

Builty, 2025-09-27:
Hi. Beta 5 is the latest one, in the releases/beta folder, updated today.

You can send CC commands also. Refer to:
https://github.com/Builty/TonexOneController/blob/main/MidiCommands.md

Midi channel is set by the web config, default is 1.
Could you double check that the Midi settings are saving? Check the web page and make sure wired Midi is enabled.

deguitarman, 2025-09-27:
maybe you can see something in this schematics: like which wire is information and which is 3.3V.
maybe the problem is here coz you never used a Midi via TRS right? only 5 pin Din to Din

[image]

so from the 5pin midi din.. out of the 5 pins, midi via TRS uses only 3 pins disregards others. See pic below

[image]

Builty, 2025-09-28:
Some users have tried it without the featherwing. It would not be compliant to the Midi standard, but some people don't care and it works well enough for them.

I'm not sure what the issue is with yours.
Perhaps you could try using the Midi channel as 4 or 6, just in case something is not matching up?

deguitarman, 2025-09-28:
it works now!

Builty, 2025-12-09:
Hi. Midi Out is enabled in the firmware versions 2.0.0.2 and above, for all platforms (except for the 4.3B.) It comes out from another pin.

For the JC3248W535, the Midi receive pin is GPIO16, and for the output/transmitter its GPIO46 (see below.)
The Featherwing pin for receive (transmit out from controller is receive in for the featherwing) is the pin right next to the normal one. Also see below. 

[image]

[image]

deguitarman, 2025-12-09:
So connect that pin on the Featherwing and connect it to the white pin on the LCD? Am I correct? hehe

[image]

Builty, 2025-12-09:
yep thats it

Builty, 2025-12-10:
You are welcome, and same to you also.