Builty/TonexOneController #275: V2 beta version on Waveshare 1.9" Touch LCD board not reading presets, "alt" light remains blue

AHMcKenzie, 2025-09-18:
Hi Builty,

I've loaded the V2 beta firmware to a Waveshare 1.9" Touch LCD board, but when I power this setup, the display won't show any presets, and the Tonex One "alt" button remains blue. Seems like the initial scan/load of preset info is not completing. 

The only pre-made firmware for this board is: TonexOneController_V2.0.0.2_beta_2_Waveshare_1_9.zip  by the name not sure this one support touch and therefore may not work with my board.

Is this the correct firmware? of not, could you point me on which source & parameters should I setup for the 1.9" touch LCD board? 
Thanks!

Builty, 2025-09-18:
Hi. The firmware you selected is the correct one. It works both with and without the touch option.

Is your Tonex pedal running the latest available firmware? So older versions are not compatible.
The Alt light is blue? That is Setup mode. Did it change to that after it connected? Normally you would have to press and hold the Alt button to get to setup mode.

I do have another user who is having issues where the Alt button doesn't change to Amber (stomp mode.) 
I'm still investigating this one.
It could be a similar thing here, or something else.
Please check the Tonex version for a start.
Thanks.

AHMcKenzie, 2025-09-18:
Thanks for your comments, I was able to make it work. My Tonex One is running the current firmware (v1.3.13), but there was definitely something going on with the power supply. My initial setup used a breadboard and didn't work, but when I connected the 5v leads directly to the S3 1.9" board pins, the Tonex box got proper power and the controller was able to read all presets! I've also connected my m-vave chocolate MIDI footswitch, which worked instantly. Very nice! thanks for making this available.

closed 2025-09-18 by AHMcKenzie

madejm, 2025-09-18:
Im happy it worked for you 😊