Builty/TonexOneController #284: ESP32-S3 4.3B crashes always, if connect the A/B-TTL-RS485-Adapter-Cables to the ESP32-S3.

Sombrero-Joe, 2025-09-29:
I'm having a problem with my Waveshare ESP 32-S3 4.3B board. I've connected ports A and B to the recommended TTL-RS485 adapter and the other side of the TTL-RS485 adapter to the MIDI Featherwing according to the wiring diagram here on Github. Whenever I connect ports A and B on the Waveshare ESP32-S3 4.3B to the TTL-RS485 adapter, the ESP32-S3 display starts to flicker, and the ESP32-S3 board crashes, with the display then going black. What could be causing this problem? Do I need to connect the ground pin (labeled with a Chinese character) on the A/B side of the TTL-RS485 adapter to GND as well (although this isn't shown in the adapter's wiring diagram)?

Builty, 2025-09-30:
Hi. Sorry for the slow reply. 
I didn't find it necessary to connect the extra ground pin. Its probably worth a try.
I'm not sure what this issue could be. Can you triple check the wiring? And maybe post a photo, in case it shows something.

Sombrero-Joe, 2025-10-01:
Thank you for the excellent support, which goes well beyond what I could have expected.
I've already checked the wiring 10 times and also tried connecting the additional ground on the TTL side. The 3.3V supply voltage is also stable. I've also already replaced the Waveshare ESP32-S3/4.3B. Unfortunately, nothing has worked so far. I've now ordered two new RS485/TTL adapters, one of which is isolated, directly from Waveshare. I'm very curious to see if this will solve the problem.
If I may, and if there's interest, I'll be happy to report on the results here after testing the two new adapters.

Sombrero-Joe, 2025-10-02:
I seem to have found and fixed the cause of the sporadic screen flickering and crashes on my Waveshare ESP32-S3 4.3B. I was the one who caused this problem. I equipped all of the ESP32-S3's inputs and outputs with two micro JST XH 2.54 8P connectors so that I could easily install and remove the Waveshare ESP32-S3 from the case without a screwdriver. I neatly coiled up the unused connector cables and secured them with cable ties. Apparently, the board's TTL inputs were interfering with the exposed cables, and the exposed cables were acting like antennas. I have now disconnected all of the unused connectors on the board. Now, there are no more crashes and no more flickering.

Builty, 2025-10-02:
Wonderful. I did have a few issues myself with long wires on the 4.3B during early dev. It doesn't seem to be quite sensitive to this.
I had forgotten about this until you mentioned it.
Great to hear its solved now.

Sombrero-Joe, 2025-10-03:
The cable between the RS484/TTL adapter and the A+ and B- connectors on the ESP32-S3 4.3B should be as short as possible and be a well-shielded, twisted-pair cable with the shield connected to GND on one side.